aredn/aredn_ar71xx #550: 5655 MHz channel 131 at 10 MHz bandwidth?

nc8q, 2020-07-30:
5, 10, and 20 bandwidths are selectable at 5920 MHz (channel 184).
Please add the selection of channels down to 130 MHz at 5 MHz bandwidth,
channel 131 at 10 MHz bandwidth, and channel 132 at 20 MHz bandwidth.
In our area of West Central Ohio, we know the location of the TDWR station.
The published frequency of that station is 5640 MHz (channel 128).
We would be using PtP links with directional antennas (dishes) and would
avoid radiating energy in the direction of the TDWR station.
I feel that we could operate at the above frequencies/bandwidths without causing interference.

Regards, Chuck nc8q

ae6xe, 2020-08-15:
Here are test images lowering the available channels from 133 to 131.   Please give a try, and submit back the support data download.  I find myself without a 5GHz node for the moment, not at primary QTH.   

(link available for a limited time:)
https://drive.google.com/drive/folders/1RlFmLf3D5U7mYYhKm_YnTr8tvtk74MXW?usp=sharing

ae6xe, 2020-08-17:
Note, "channels down to 130 MHz at 5 MHz bandwidth"  would be outside of US part 97 lower allocation of 5650.   for a channel center at ch 130 (5650) @ 5MHz channel width, there would be 2.5Mhz of bandwidth below allocation.

PR #551 (2020-08-18): aredn: extend 5GHz channel options from 133 down to 131
In the US, part 97 ham radio lower band edge is 5650.
Channel 131 @ 10MHz channel width is the lowest channel
setting that stays within allocation.

fixes #550
commit 6e579d4d (2020-08-18): aredn: extend 5GHz channel options from 133 down to 131

In the US, part 97 ham radio lower band edge is 5650.
Channel 131 @ 10MHz channel width is the lowest channel
setting that stays within allocation.

fixes #550
closed 2020-08-18 by ae6xe