aredn/aredn_ar71xx #572: RBLHG-5HPnD-INTL !!!! UNSUPPORTED DEVICE !!!!

gelmce, 2020-09-24:
May be another Mikrotik device name change.
https://www.arednmesh.org/content/rblhg-5hpnd-intl-unsupported-device

Yet another boardid ?

It seems that the LHG-5HPnD-XLs are scarce.
I found a US retailer that had some INTL units available
(After submitting their release and showing my FCC license).

Is this a new boardid ?

node: NOCALL-37-46-81
model: MikroTik RouterBOARD RBLHG-5HPnD

!!!! UNSUPPORTED DEVICE !!!!
boardid: MikroTik RouterBOARD RBLHG-5HPnD
Device has not been tested.
Please file a ticket with your experiences.

br-lan 48:8F:5A:26:2E:51
eth0 48:8F:5A:26:2E:51
eth0.1 48:8F:5A:26:2E:51
eth0.2 48:8F:5A:26:2E:51
tunl0 00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0 48:8F:5A:25:2E:51

/proc/cpuinfo
system type : Atheros AR9344 rev 3
machine : MikroTik RouterBOARD RBLHG-5HPnD

Chuck

ae6xe, 2020-09-25:
Yes, new board id.  Try these test images for this device:

https://drive.google.com/drive/folders/1PhjMYoOBv1t-_OFGC6HS3BEdDR77NWqE?usp=sharing

gelmce, 2020-09-28:
Trying to load AREDN on new RBLHG-5HPnD-XL-US

Loaded NB-1819.elf
<i>
root@W6CDR-LHG5HP-XL-111:~# cat sysinfo/board_name 
unknown
root@W6CDR-LHG5HP-XL-111:~#
</i>
ssh'ed into the node and changed sysinfo/board_name from unknown:
<i>echo "rb-lhg-5hpnd" > sysinfo/board_name</i>
Loaded NB-1819.bin
Channels were shown by frequency. Default was '5.180 GHz'
Set nodename and password...save...rebooted.
Booted up with NB-1819...channels still shown as frequency.
-----
 node: W6CDR-LHG5HP-XL-111
model: MikroTik RouterBOARD RBLHG-5HPnD

!!!! UNSUPPORTED DEVICE !!!!
boardid: MikroTik RouterBOARD RBLHG-5HPnD
Device has not been tested. Please file a ticket with your experiences.

br-lan 48:8F:5A:4B:75:6F
eth0   48:8F:5A:4B:75:6F
eth0.1 48:8F:5A:4B:75:6F
eth0.2 48:8F:5A:4B:75:6F
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  48:8F:5A:4A:75:6F
wlan0-1 4A-8F-5A-4A-75-6F-00-44-00-00-00-00-00-00-00-00

/proc/cpuinfo
system type		: Atheros AR9344 rev 3
machine			: MikroTik RouterBOARD RBLHG-5HPnD
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 299.82
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit nan_legacy nan_2008 perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='48:8f:5a:4a:75:6f'
hsmmmesh.settings.mac2='74.117.111'
hsmmmesh.settings.dtdmac='74.117.110'
hsmmmesh.settings.node='W6CDR-LHG5HP-XL-111'
hsmmmesh.settings.config='mesh'

Part of the AREDN™ Project. For more details please see here
-----

When you get around to it, ;-)