aredn/aredn_ar71xx #578: NanoStation M2 XM - Unsupported on ATH79 FW

xJMV, 2020-10-05:
I've tried to update this NSM2 from 3.20.3.0 to 3.20.3.1. Once rebooted, it is not responding unless connecting to secondary port to see the unsupported banner. This unit was working for few weeks now in a test setup.

Included are support files (re-zipped so it can be uploaded here).
[supportdata-NOCALL-202005161837.zip](https://github.com/aredn/aredn_ar71xx/files/5326483/supportdata-NOCALL-202005161837.zip)

ae6xe, 2020-10-06:
xJMV, this might be a corruption of the flash on your device.    Ubiquiti devices have a "boardid" that uniquely defines the model.  This is the code that pulls the boardid off the flash:

  if [ "$(/usr/local/bin/get_hardware_mfg)" = "Ubiquiti" ]; then
    [ -f "/sys/devices/pci0000:00/0000:00:00.0/subsystem_device" ] &&
            name="$(cat /sys/devices/pci0000:00/0000:00:00.0/subsystem_device)"
    [ "$name" = "0x0000" -o -z $name ] &&
            name="0x$(dd if=/dev/mtd7 bs=1 skip=12 count=2 2>/dev/null | hexdump -v -n 4 -e '1/1 "%02x"')"
  fi

On your device this is coming back with "0xffff", when it should be "0xe012".   Are you familiar with how to telnet to the device and do a vi command to edit a file?   If so, in the file /www/cgi-bin/perlfunc.pm , find the string "0xe012" and change it to "0xffff", reboot and see what happens.   For the string it does find, this will force it to be a nanostation M2 XM.    ...other things may not behave correctly, if indeed there is a flash issue and wider corruption.

Joe AE6XE

ae6xe, 2020-10-06:
Note, I'm assuming that this is not reproducible on other NSM2 XM devices.    It would be unusual to have a release go this long and no one else see this, should it be a defect that got through.  Best that we confirm this...

Joe AE6XE

xJMV, 2020-10-07:
> 
> 
> Note, I'm assuming that this is not reproducible on other NSM2 XM devices. It would be unusual to have a release go this long and no one else see this, should it be a defect that got through. Best that we confirm this...
> 
> Joe AE6XE

I do not have another NSM2 XM.... I got it has a "gift" and use it to link test rigs to home hAP.

I will test your changes this week. I remember having trouble flashing this one from ar71 to ath79, IIRC I had to use bullet firmware to make it works...

xJMV, 2020-10-07:
@ae6xe it "works". It do not display the unsupported banner anymore... But still no joy for networking and computing the IP value (getting 10.0.0.0).

Reverted to 3.19 and no joy anymore... Guess it is time to let it go to heaven! Thanks for your time!

closed 2020-10-07 by xJMV