aredn/aredn_ar71xx #579: [Enhancement] Include the HEADING information of the nodes with the coordinates

kp4msr, 2020-10-07:
it will be good to have that information in the systeminfo.json API and in the User Interface of the node, a field to select or write what is the direcction of the node, for example: HEADING NORTH, OR SOUTH EAST, OR EAST, OR WEST... that could help the other users to check which antenna its reachable from his location. .

r1de, 2020-10-09:
Should it be a numerical heading or a Cardinal heading?
270 vs. WEST, 90 vs SOUTH, 135 vs SOUTHEAST, etc...
(and don't even get started on "North-Northwest" and "Northeast by North" and all that!)

How "accurate" should it be? Exact numbers or a general direction via some dropdown menu?

xJMV, 2020-10-09:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Numerical would be awesome for mapping purpose and more precise for calculation on software like radiomobile.                                                                                                                                                                                                                                                                                                                From: notifications@github.comSent: October 8, 2020 22:15To: aredn_ar71xx@noreply.github.comReply to: reply@reply.github.comCc: subscribed@noreply.github.comSubject: Re: [aredn/aredn_ar71xx] [Enhancement] Include the HEADING information of the nodes with the coordinates (#579)  
Should it be a numerical heading or a Cardinal heading?
270 vs. WEST, 90 vs SOUTH, 135 vs SOUTHEAST, etc...
(and don't even get started on "North-Northwest" and "Northeast by North" and all that!)
How "accurate" should it be? Exact numbers or a general direction via some dropdown menu?

r1de, 2020-10-09:
> Numerical would be awesome for mapping purpose and more precise for calculation on software like radiomobile.
  
Very true! And it would be much easier to code, just a new text box. :smile:   
As the "mapping dude" I'll give this a :+1: as I think it might be useful.

xJMV, 2020-10-09:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  While adding heading... Why not elevation above ground, antenna gain and beam width?I usually add those into comments for references but having all them in parameters that could be fetch from sysinfo.json would allow automation and maybe even display coverage in mesh map...                                                                                                                                      Just saying...                                                                                                                                                                           From: notifications@github.comSent: October 8, 2020 22:24To: aredn_ar71xx@noreply.github.comReply to: reply@reply.github.comCc: Xop777@gmail.com; comment@noreply.github.comSubject: Re: [aredn/aredn_ar71xx] [Enhancement] Include the HEADING information of the nodes with the coordinates (#579)  

Numerical would be awesome for mapping purpose and more precise for calculation on software like radiomobile.

Very true! And it would be much easier to code, just a new text box. 😄

As the "mapping dude" I'll give this a 👍 as I think it might be useful.

kp4msr, 2020-10-09:
Agreed, thats our vision. I am developing an app for aredn and i need that data.

The standard is elevation and azimuth i guess.
Commets are not the best place for this, it should have it owns fields for that kind of info and visible thru de API.

What do you think ?

r1de, 2020-10-09:
That is not a bad idea either. I agree!  
It would not be bad for future development to have all that in the sysinfo.json file.   
The mention of [RadioMobile](http://www.ve2dbe.com/english1.html) makes it's even nicer to have, I forgot about that program!  
  
I know I have said this and it has bitten me in the arse, but adding this should not be hard at all.  
Just have to find where to put it, I guess down by the location info.  
It will of course still be optional to fill it out.  
    
I can't wait to see what other maps people come up with!  
  
I will start working on this, it can be assigned to me.

kp4msr, 2020-10-09:
I am creating a aredn network for the entire Puerto Rixo island, i have lot of ideas on my mind and i am software engineer.  If we meet and we walk thru the source code i can contribute with those and other changes.

We are planning to fill the island of aredn antennas first on sites and tops of mountains, then in people houses, hospitals, goverment agencies and so on. We will provide free internet to everybody with limited bandwith for send emails, and voice ip phones to each one.

Its a big proyect, but as technical lead and software engineer and the people involved in this project it could be possible.

MS

kp4msr, 2020-10-11:
Please let me know if you will implement or if implemented already, i can collaborate if required.

The best way should be azimuth and elevation using numerical degrees values. 
I cant wait to have this info available in the systeminfo.json API, to implement a feature in our application.

r1de, 2020-10-14:
If you can create a new PR for it, that's the best way! :smile:  
For me, life is getting in the way right now and will probably continue to do so for the foreseeable future, I am not sure when I will get to it at this point.

xJMV, 2020-10-14:
@r1de just for info whenever you get back...

This is actually what I do with nodes... Beam is beam "angle". So it becomes easy chart a shape around a node when someone click on it :)
[image]

Take care!

r1de, 2020-10-16:
That's good. A lot of people use that method.  
I do agree though, that it would be nice to have that info in sysinfo.json (when said info is available).  
It would be easier to use programmatically that way.  
  
I am not sure when I will be able to get to it tho. I take back the "assign it to me" statement I made earlier.  
Real Life is definitely getting in the way.  
  
It shouldn't be hard, any of you can add it! Just try it! :smile: