aredn/aredn_ar71xx #582: RBLHG-2nD !!!! UNSUPPORTED DEVICE !!!!

dfujikiv, 2020-10-18:
New MikroTik RBLHG-2nD force loaded Nightly Build aredn*.bin file resulted in a partially functional load with "UNSUPPORTED DEVICE" header and missing channels below channel 1.

Here is the web sysinfo data:
============================================
http://localnode:8080/cgi-bin/sysinfo
============================================
 node: KE0NA-MSP-LHG
model: MikroTik RouterBOARD RBLHG-2nD

!!!! UNSUPPORTED DEVICE !!!!
boardid: MikroTik RouterBOARD RBLHG-2nD
Device has not been tested. Please file a ticket with your experiences.

br-lan 48:8F:5A:7B:73:D4
eth0   48:8F:5A:7B:73:D4
eth0.1 48:8F:5A:7B:73:D4
eth0.2 48:8F:5A:7B:73:D4
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  48:8F:5A:7A:73:D4
wlan0-1 4A-8F-5A-7A-73-D4-00-44-00-00-00-00-00-00-00-00

/proc/cpuinfo
system type		: Qualcomm Atheros QCA9533 ver 2 rev 0
machine			: MikroTik RouterBOARD RBLHG-2nD
processor		: 0
cpu model		: MIPS 24Kc V7.4
BogoMIPS		: 432.53
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 16
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit nan_legacy nan_2008 perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='48:8f:5a:7a:73:d4'
hsmmmesh.settings.mac2='122.115.212'
hsmmmesh.settings.dtdmac='122.115.211'
hsmmmesh.settings.node='KE0NA-MSP-LHG'
hsmmmesh.settings.config='mesh'

I zipped the TGZ support file and attached it here:

[supportdata-KE0NA-MSP-LHG-202005162018.zip](https://github.com/aredn/aredn_ar71xx/files/5397990/supportdata-KE0NA-MSP-LHG-202005162018.zip)

_
Thank you for your help,
Dave - KE0NA

ae6xe, 2020-10-18:
Look for images in the next nightly build  to address this issue.   Code has been submitted that knows about this new model string showing up in recent Mikrotik LHG 2nD devices.

dfujikiv, 2020-10-18:
Thank you Joe!

dfujikiv, 2020-10-18:
Thanks Joe, your efforts are much appreciated!

Dave, KE0NA

PR #583 (2020-10-18): AREDN: add new Mikrotik RBLHG-2nD model string
fixes #582
commit bf944dfa (2020-10-18): AREDN: add new Mikrotik RBLHG-2nD model string

fixes #582
closed 2020-10-18 by ae6xe