aredn/aredn_ar71xx #585: Support for Mikrotik SXT-5HPnD or SXT-G-5HPnD-SA r2?

Rendezvousca, 2020-11-23:
I have a couple of the above Mikrotik devices laying around I was trying to get up and test. They are several years old c.2014. I am unable to successfully load the .elf file with windows machine. 

Running the PXE server on windows, the devices boot up and pull the rc.elf image successfully. I see the “ TFTPd: DoReadFile: rb.elf”. After that I am not able to connect to the device. I changed the network adapter back to DHCP but it does not pull an IP address. It just assigns a link-local 169.254 address. I tried statically assigning the 192.168.1.10 address but still cannot reach the admin page. IP scanner shows nothing. No ping to 192.168.1.1

It doesn't appear to be in a “boot loop” like some of the other issues Ive read about. Its not cycling power or network connection. 

Sorry if this is the wrong place to ask this. I dont have access to the forum yet.

xJMV, 2020-11-23:
Once you have sent the .elf, wait for it to reboot. Keep your network static to access the GUI and upload the .bin file. it will reboot again. Configure it (hostname and password). It will reboot again. During that last reboot you will set your network interface in DHCP mode again.

Until it is configured the node will keep its IP as 192.168.1.1

ae6xe, 2020-11-23:
There are some older 6 sided hex shape SXT models.   These models  didn't work with openwrt and not ported to work for aredn.  The models working in openwrt and aredn are square shaped cases.

Rendezvousca, 2020-11-23:
Thanks Joe. Indeed these are the older hex shape SXT’s.

Thanks

closed 2020-11-23 by Rendezvousca